atomic14/esp32-walkie-talkie #1: Simple code for sending audio that use MAX9814

rachanon0807, 2021-04-06:
I want to know how to use simple code for sending audio that speak through microphone MAX9814

Can you please check my code

`#include <driver/i2s.h>
#define I2S_SAMPLE_RATE 16000
#define ADC_INPUT ADC1_CHANNEL_7 //pin 35
#include <esp_now.h>
#include <WiFi.h>

uint8_t buff_count = 0;
#define SPKR_BUFF_SIZE 60
uint32_t DataTransmit[SPKR_BUFF_SIZE];
uint8_t broadcastAddress[] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

bool send_success;

void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
 
}
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {

}

void setup() {
  pinMode(27,OUTPUT);
  // put your setup code here, to run once:
  Serial.begin(115200);
  esp_err_t err;
  const i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  I2S_SAMPLE_RATE,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_LSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 64,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
  };
 err =  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
   i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);
  i2s_adc_enable(I2S_NUM_0);
  
  
 

WiFi.mode(WIFI_STA);

 
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  
  esp_now_register_send_cb(OnDataSent);
  
  
  esp_now_peer_info_t peerInfo;
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;

 if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
  esp_now_register_recv_cb(OnDataRecv);
}

void loop() {
  int32_t sample = 0;
   while( buff_count!=SPKR_BUFF_SIZE-1 ){
  int bytes_read = i2s_pop_sample(I2S_NUM_0, (char *)&sample, portMAX_DELAY);
  Serial.println(sample);
  if (bytes_read > 0) {
      DataTransmit[buff_count] = sample;
      buff_count++;
    }
   }
   buff_count = 0;
     esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &DataTransmit, SPKR_BUFF_SIZE*4);

}`

cgreening, 2021-04-06:
There are some quite weird things with the built-in ADCs - I've got a video here: https://www.youtube.com/watch?v=pPh3_ciEmzs

And you might want to check the sample code here: https://github.com/atomic14/esp32_audio/blob/master/i2s_sampling/src/ADCSampler.cpp

In particular, you need to pre-process the samples from the ADC with the following formula:

```
(2048 - (rawSamples[i] & 0xfff))
```
Only the bottom 12 bits are relevant and the data is inverted.

You're using the i2s_pop_sample function - check the docs for this: https://docs.espressif.com/projects/esp-idf/en/v3.3/api-reference/peripherals/i2s.html

    sample: Buffer to write data. Size of buffer (in bytes) = bits_per_sample / 8.

So your line of code:

```
int32_t sample = 0;
```

Should be

```
int16_t sample = 0;
```

rachanon0807, 2021-04-06:
Okay, thank you very much for the information. You made me understand a lot, I'll give it a try.

closed 2021-10-04 by cgreening