atomic14/esp32-walkie-talkie #5: how to use code

MichalZima, 2021-06-19:
hello, I got into troubles when I tried to use your code and library. as it's not made as other library (just code with examples demonstrating functions of library, that you can #include in arduino IDE code). I am a bit lost in the Platformio IDE and way, how your code should be used properly. Could you explain how to use this library and how to use Platformio please? I tried to search for such a video on your ytb channel but I haven't find any yet. If you have any, it would be awesome. Can you send me a link to your or someone else's video, or some other source of information (if you know about any) that explains this issue? It would be a great help.

Thank you very much.

cgreening, 2021-10-04:
Unfortunately, the code isn't really designed to be used as a library - it's a stand-alone application. If you open it in vscode with the platform io plugin installed then you should be able to build and flash it onto your device.

FedericoBusero, 2021-10-10:
I did a pull request to make compilation a bit easier on other ESP32 chips, but also it makes it possible to compile it in the Arduino IDE. See https://github.com/atomic14/esp32-walkie-talkie/pull/14/files

nielsnl68, 2021-10-14:
Hi, sorry to jump into this conversation, i have been playing around with ESP-NOW for a while now. And i created my own protocol above ESP-NOW named ESP-TALK, Just to do the same thing as you suggested @MichalZima. The only part that is still missing is audio part. You can find it here on github. 

Although the broadcast option of ESP-NOW sounds very nice. It is in real live not so friendly because every device that is using ESP-NOW will get our message. And when they did not bother to add a header to there message they and you will get lots of interference from each other.

The best solution imho is just talk to the nearest repeater and the repeater should send to each walky-talky and other repeater's in his network. 

Good luck and i hope to see some results from you in the near future.

JorgeOO, 2021-10-17:
Good afternoon,

Congratulations for the project.
Can you guide me to know what to modify if I change the microphone to 441?
Yours is out of stock.

Thanks.

FedericoBusero, 2021-10-17:
You don´t need to change anything to use the microphone  INMP441, as this is a standard I2S microphone.

cgreening, 2022-08-26:
Closing this due to lack of activity.

closed 2022-08-26 by cgreening