atomic14/esp32-walkie-talkie #17: Configuration with INMP441 microphone and DAC

lcaum, 2021-11-04:
Hello,

I'm trying to make this walkie-talkie code work with another configuration:
On one side I have an ESP32 emitter with a INMP441 microphone.
On other side I have another ESP32 receiver with ESP32 buildin DAC + amplifier.
The ESP_NOW link seems to work well.
My issue is the sound is very noisy. The sound changes when I make some noise near the microphone but with some sort of saturation effect and with time I have something like an acoustic feedback effect. 
Do you have an idea how I could fix this ?
Thanks

lcaum, 2021-11-06:
Hello,
I tried other code from https://github.com/atomic14/esp32_audio and I was able to validate that output and input circuits are working fine.
Finaly, I changed the transport to standard WIFI and it works much better.
It seems something goes wrong with ESP Now.

AdamSwiercz, 2021-12-11:
Hello,
if I try to NOT use INMP441 microphone, I commented line like this:
`// #define USE_I2S_SPEAKER_OUTPUT` 
but if I do so I get error
`Compiling .pio\build\lolin32\lib809\audio_input\I2SSampler.cpp.o
src\Application.cpp: In constructor 'Application::Application()':
src\Application.cpp:39:18: error: expected type-specifier before 'DACOutput'
   m_output = new DACOutput(I2S_NUM_0);
                  ^
Compiling .pio\build\lolin32\lib310\audio_output\DACOutput.cpp.o`

What I should to do with the code to have analog audio on GPIO25?

FedericoBusero, 2021-12-11:
Add
`#include "DACOutput.h" `
to Application.cpp?

AdamSwiercz, 2021-12-11:
Thanks for reply,
I was trying and:

`Compiling .pio\build\lolin32\lib152\WiFi\WiFiAP.cpp.o
src\Application.cpp: In constructor 'Application::Application()':
src\Application.cpp:39:12: error: cannot convert 'DACOutput*' to 'I2SOutput*' in assignment
   m_output = new DACOutput(I2S_NUM_0);
            ^
*** [.pio\build\lolin32\src\Application.cpp.o] Error 1`

FedericoBusero, 2021-12-11:
Replace I2SOutput by Output in Application.h?

AdamSwiercz, 2021-12-11:
Cannot say if it worked, but it compiling now without errors.
 this is what Application.h now consist:
`#pragma once

//class I2SOutput;
class I2SSampler;
class Transport;
class OutputBuffer;
class IndicatorLed;
class Output;

class Application
{
private:
  //I2SOutput *m_output;
  I2SSampler *m_input;
  Transport *m_transport;
  IndicatorLed *m_indicator_led;
  OutputBuffer *m_output_buffer;
  Output *m_output;

public:
  Application();
  void begin();
  void loop();
};`

Thank you Federico. :)

FedericoBusero, 2021-12-12:
I have fixed the compiling issues in the library and tested that it works using the DAC output.
Remark that the sound is weak, it is better to increase the gain by changing 5 into 8 in OutputBuffer:
change
`        samples[i] = (sample - 128) << 5;`
into
`        samples[i] = (sample - 128) << 8;`

cgreening, 2022-08-26:
Looks like this was solved by @FedericoBusero

closed 2022-08-26 by cgreening