atomic14/esp32-walkie-talkie #18: Instructions for use with ADC microphone and DAC headphones

Bartvelp, 2021-11-28:
Hi,
On most pages you explain you can use analog components instead of I2S ones, I'd like to use this project without the I2S components. In your youtube video and hackaday I can see you using a pair of headphones.
[image]

I think you are only using it as output since you are combining left and right audio, and nothing is connected to the microphone (sleeve).
I want to use the built in analog microphone as well, since that would reduce components to what most hobbyists already have. I think you need a bias voltage though, since there is a FET inside that needs to be powered. Do you have any idea on whether this is possible, and if so how to wire it?
Thanks for the cool project

cgreening, 2021-11-29:
I have previously had a play with trying to use the microphone in the apple headsets and didn't have much success.

It should be possible to use with the correct preamp and power supply, but I haven't found any instructions on how to do it.

Sorry I can't be more help as it would be a very nice hack,

closed 2021-11-29 by cgreening

nielsnl68, 2021-11-29:
As what Chris said. I have seen some youtube vid's trying this approche with a good sound return.

cgreening, 2021-11-29:
If you’ve got any links with good instructions that would be awesome as I’d
love to get it working.

nielsnl68, 2021-11-29:
Lol, basically it was your vid i was talking about. Try to play MP3 files.

Bartvelp, 2021-11-29:
Yes sound output is quite easy (?) But mic input is harder.
I found this link but it is not very complete https://electronics.stackexchange.com/a/547636/223026
I understand almost nothing about AC so maybe I am using the wrong search terms

cgreening, 2021-11-29:
I’m theory, it should just be case of supplying a some small voltage on the
mic wire and then feeding it into an AC couple amplifier. But I think the
apple headsets have a bit of “handshaking” with the device to activate them.

Some people say you can hold down the button when plugging them in to make
them work with android devices - but there’s not much info out there.

Bartvelp, 2021-12-01:
What makes you think Apple headset require handshaking?
Also I'm open to using other headsets, there are a lot of cheap headphones out there which could work just fine since they adhere to the "Apple" TRRS spec.

I will look into it when I got time, but my oscilloscope broke on me so it's hard to debug.