atomic14/esp32-walkie-talkie #21: Question: possible to use this code with a serial connection

Sennevds, 2021-12-15:
Hi,

I really like this project and I have used it with esp-now and works okay. Now I want to use this in a house where there are old intercoms and bad reception :). I thought I maybe could reuse the original wiring and instead of esp-now use wired connection between esp's.
Would this be possible?

FedericoBusero, 2021-12-16:
A possibility is to use rs-485 for example with max485. You would have to write a new transport class for it. 

Or you can simply try to use the wires as antenna wire to transport the esp-now wifi signal.

But did you first try to use an ESP with an external antenna?

Sennevds, 2021-12-17:
I didn't test with external antenna because I had hoped that with cables it would be easier :) I will check how easy it is to add an external antenna to the Lolin32. Thanks for the info!

closed 2021-12-17 by Sennevds