atomic14/esp32-walkie-talkie #25: Using DAC ouput to amp instead of I2S

griffinmt, 2022-10-26:
It is not clear (to me at least) what I need to do or set up to have the audio output use DAC instead of I2S. I am using a PAM8302 with attached volume control and running ESP-NOW. Any and all help would be appreciated.

martyn.t.griffin@gmail.com

griffinmt, 2022-10-26:
> The code is not currently set up to output to the DAC, but o think to
> should be straightforward to make it work. I'll have to have a look.

I would truly appreciate it.

cgreening, 2022-10-26:
Looks like it's already there!

in the `config.h` file, just comment out the line:

```
#define USE_I2S_SPEAKER_OUTPUT
```

griffinmt, 2022-10-26:
Already tried that but no change. I tried commenting the other 4 i2s
speaker lines, but that just raise errors.

griffinmt, 2022-10-27:
your description states the following:
" modify the code to use the built-in DAC for output "

cgreening, 2022-10-31:
That should be all that's necessary - that will start writing samples out to the DAC.

Might be worth running a simple test to make sure you can output audio normally to the DAC and sound comes out.

finlaydegrauwe, 2023-01-23:
I've been trying to achieve the same, but so far no result. I think it's something simple with defining the pin of the built-in DAC somewhere (GPIO 25 or 26), but I can't seem to figure out where to put this.