atomic14/esp32-walkie-talkie #26: Received audio is only static noise

ericlindersson, 2023-01-29:
I have connected everything according to the code, default pin numbers et.c. I can see that data is being received from the mic and being sent. The receiver receives data, but I can only hear a ticking noise from the speaker. I have also tested the components alone and was able to record my voice over UDP, and used the speaker with a webradio sample with success.
I have only tries ESP-NOW, could that be the problem? Or is there anything else I could try?

meloluan, 2023-03-30:
I have the same problem, did you manage to solve it?

FedericoBusero, 2023-05-22:
The code was not yet compatible with IDF version 4, now it should work again.

In case the microphone is not yet recording audio, in some versions on some chips you need to set 
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_RIGHT
instead of left.

Remark also that defines can be set in platformio.ini or config.h  : USE_I2S_MIC_INPUT & USE_ESP_NOW are set in platformio.ini