atomic14/esp32-walkie-talkie #29: How hard would it be to add targeted transmission?

AussieMakerGeek, 2023-03-06:
Hi, I'm wanting to build a multi-room Intercom and this project is perfect but would broadcast to all units. I want to be able to target specific room-to-room intercom but also still have a broadcast mode.

I was thinking that perhaps there could be a UDP packet header to target a specific 'ID' based on which transmit button was pushed. Then certain receivers would only play back sound targeted with their ID, but still have a '00' ID for a broadcast? Could this work?

I am willing to put the time into the code, but just not sure where to start, so if you could point me in the direction i'll give it a crack - I'm usually coding in Arduino so platformIO is not native to me and a tad daunting but willing to learn it for this project.

AussieMakerGeek, 2023-03-07:
So, to partially answer my own question now that I have gone over the code better, I can see that I can use the `set_header` function in the code, based on whatever button I push (TBD) to set the header on the outbound broadcast but where in the receive side does it verify the header? I am intending on using UDP only, not ESP_NOW as the devices will be too far away to support peer=to-peer networking.

rhambus, 2024-11-10:
@AussieMakerGeek Did you ever get this working? I would really like to set up a multi room intercom system as well.

AussieMakerGeek, 2024-11-11:
In short I did, however, I could not achieve the desired voice quality so I sort of abandoned the project. That's not to say the code was no good, it was probably my hardware implementation during testing (breadboard and jumpers) that was the limiting factor. But it was legible, and I was able to direct voice to specific endpoints.

I'm pretty sure my [fork ](https://github.com/AussieMakerGeek/esp32-Multi-room-intercom) of this repo has the updated code where I got it to work, but it was some time ago and I don't recall if I updated the final code here. If not, I'm sure I have it locally.