atomic14/esp32-walkie-talkie #37: Help needed.

LieutenantThanon, 2023-09-05:
Hello, I'm a beginner to C++. My question is how can I edit the code to adapt for use with the following components :
MAX9814
PAM8403
And also for the NRF24L01 Antennas as I may need the extended range.

Thanks in advance.

FedericoBusero, 2023-09-05:
It looks like both microphone and amplifier are analogue.
So USE_I2S_MIC_INPUT and USE_I2S_SPEAKER_OUTPUT may not be defined (check platformio.ini and config.h). 

But the audio quality is maybe better when using i2s modules instead of analogue.

The microphone pin is defined by 
// Analog Microphone Settings - ADC1_CHANNEL_7 is GPIO35
#define ADC_MIC_CHANNEL ADC1_CHANNEL_7

The analogue output pin is fixed in the chip. (DAC).

Your module NRF24L01 cannot be used in this project, I think. For more info on extending the range of ESPnow, check youtube from the creator of this project or from the guy with the swiss accent.

Anyhow, if you need more help, you need to tell which chip you are using exactly.

LieutenantThanon, 2023-09-05:
The board I am currently using is the 38 pin NodeMCU-32S.

FedericoBusero, 2023-09-05:
This should work the way I described in my previous post

Platformio.ini:

[env:nodemcu-32s]
board = nodemcu-32s
platform = espressif32@3.4.0
build_flags = -Ofast -D USE_ESP_NOW
lib_ignore = indicator_led_pico

In config.h: delete folowing line
#define USE_I2S_SPEAKER_OUTPUT

Then you have
Analog out on D25 and/or D26
Analog in on D35
Button on D23