atomic14/esp32-walkie-talkie #39: Migrating to M5 Cardputer

Kabron287, 2024-01-24:
Hello,
I made changes in **platformio.ini** and in **Application.cpp**(GPIO_TRANSMIT_BUTTON 0 and active LOW, I2S pins also redefined) to work with m5stack-stamps3:
 [env:m5stack-stamps3]
board = m5stack-stamps3
platform = espressif32
build_flags = -Ofast -D USE_I2S_MIC_INPUT -D USE_ESP_NOW
lib_ignore = indicator_led_pico

 in **platformio_override.example.ini**:

[env:esp32dev]
;;board = esp32dev
board = m5stack-stamps3
build_flags = -Ofast -D USE_ESP_NOW -D USE_I2S_MIC_INPUT

Compilation OK but serial out only:

ESP-ROM:esp32s3-20210327                                                        
Build:Mar 27 2021                                                               
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)                  
Saved PC:0x4202173e                                                             
SPIWP:0xee                                                                      
mode:DIO, clock div:1                                                           
load:0x3fce3808,len:0x44c                                                       
load:0x403c9700,len:0xbe4                                                       
load:0x403cc700,len:0x2a68                                                      
entry 0x403c98d4  

and nothing more.
What else have I  change to?

Kabron287, 2024-01-25:
I have some progress:
noise on one unit while transmit on another one.

Definitelly something wrong with mic settings:
#define I2S_MIC_CHANNEL I2S_CHANNEL_FMT_ONLY_LEFT
#define I2S_MIC_SERIAL_CLOCK 43     
#define I2S_MIC_LEFT_RIGHT_CLOCK -1 
#define I2S_MIC_SERIAL_DATA 46     

i tested all other possible I2S_MIC_CHANNEL values with no success.

What else could be done for SPM1423HM4H-B mic?

FedericoBusero, 2024-01-25:
I want to test S3 next week, you can test

platform = espressif32@6.3.1

I made some i2s test programs for s3 already, on
https://github.com/Fri3dCamp/badge_2024_arduino/tree/main/examples/platformio%20basic%20examples

Kabron287, 2024-01-29:
Well, I did it.
BUT, there are nuances.
1. SPM1423 mic used PDM coding. After makig corresponding correction in config.h and config.cpp, sound did not appeared and I did not realize why. So, to make it work I have to use mic part of M5Cardputer library. Now it works fine.
2. Code, compiled with platformio won't work at all, so, I have to adapt it for Arduino.
With these synthetic modification I got it work.