atomic14/esp32-walkie-talkie #40: it is not an issue: how to update for a full duplex telecom

philippedc, 2024-02-19:
Hi, first thank you for your job, it is so well explained!

I would like to build a full-duplex version (without the push-button) therefor it is not so easy. Because the push-button makes the device either in receiver or transmitter mode.

Have you got an idea how to add this feature? I suppose both modes can't be done at the same time....

SumayyahAli, 2024-03-26:
Dose this version of code work with you now  or your faced some compatibility issue ?

philippedc, 2024-03-26:
Hi @SumayyahAli, I did not test this code. What I want to build are:
- a kind of baby monitor with esp32-CAM and audio
- a doorbell video monitor with full duplex audio.
I suppose it should not be very hard to add a picture from time to time to the audio, as I do not need a continuous video stream. But there are not many examples to help for the project. And for the moment I have not time to study VOIP nor mp3 to avi converter.

FedericoBusero, 2024-03-26:
Maybe this project is not the ideal start of your project. Did you already check https://github.com/s60sc/ESP32-CAM_MJPEG2SD or https://github.com/jameszah/ESP32-CAM-Video-Recorder ?

Also regarding hardware, there is currently better hardware such as the Xiao ESP32S3 Sense

philippedc, 2024-03-26:
Many thanks @FedericoBusero, I will test https://github.com/jameszah/ESP32-CAM-Video-Recorder 
About the s60s project, it does not work for me as I have a fake ESP32 CAM with 2Mb PSRAM only when 4Mb is required.